UofA-SPEAR/pcb_designs #47: Subsystem PDU re-design

davidlenfesty, 2021-09-12:
The subsystem PDU is mostly functional, however it needs a few fixes before it's "production ready".

Required Improvements:
- Reverse XT60 connectors to correct their orientation.
- Add LED's to indicate input power and each output channel (ideally different colours, maybe green on input and red on output?)
- Fix thermal relief sizes on copper pour, they are currently much too small.

@GV-Singh2000 Please install EAGLE (you can get a student license with your ualberta email, although the process is a bit convoluted right now, I think you have to install Fusion 360 first? it's weird.) and add this repo to your projects search directory. If you're not sure we can organize a time to go over it together and go over the board as well.

GV-Singh2000, 2021-09-12:
OK, I will start by installing EAGLE or Fusion 360 Firs and then I guess I might need some help as I have no idea what the above things really mean, Its Embarrassing but I will try my Best to understand and work on them

davidlenfesty, 2021-09-12:
All good, we'll go over it in person then, I'll figure out when I have time this week (labs all start so it will be a bit hectic for me.

GV-Singh2000, 2021-09-12:
Sure, Thank you its same for me, I have 4 labs every week from Monday to Thursday 2-5 PM.